Infineon/TLE493D-3DMagnetic-Sensor #7: Cartesian_low_power example always showing same data

mlvandenbos, 2019-04-01:
When trying the library with a '3D magnetic sensor 2GO' board, with example 'Cartasian_low_power' is always shows this in serial monitor:

[image]

I'm using the arduino IDE v1.8.5 and library v1.0.1.
The board itself is working fine in the 3D toGo IDE.

The output of the compiler/uploader:

```
--------------------------
Infineon XMC Flasher Lite
Copyright Infineon Technologies 2017
--------------------------
Windows 10
Operating System: Windows 10

Initialisation
--------------------------
Windows 10
Loading JLink Library... 
JLink Library Version: V6.44d

Configuring Device
Device Name : XMC1100-0064
Interface   : SWD
Speed       : 4000
File Upload
--------------------------
Uploading C:\Users\xxxxxx\AppData\Local\Temp\arduino_build_506221/Cartesian_low_power.ino.hex
... Done
--------------------------
... Finished Succesfully
```

jaenrig-ifx, 2019-04-02:
Hi mlvandenbos,

Could you let us know the specific hardware board so we can try to reproduce the issue? There are two variants: 
https://www.infineon.com/cms/en/product/sensor/magnetic-position-sensor/3d-magnetic-sensor/
[image]

Regards,
Juan A

mlvandenbos, 2019-04-02:
According the GUI application I have a TLE493D A1B6

[image]

jaenrig-ifx, 2019-04-02:
Hi mlvandenbos,

This library is for the TL**E**493D sensors family. There is an specific library for the TL**V**493D variant in this [repository](https://github.com/Infineon/TLV493D-A1B6-3DMagnetic-Sensor).

Best Regards,
Juan A

mlvandenbos, 2019-04-02:
Hi Juan,

I'm sorry. I completely overlooked the E/V difference.

I will try that library.

closed 2019-04-02 by mlvandenbos

jaenrig-ifx, 2019-04-02:
No problem. It is too similar not to be misleadin :)
Let us know if you experiment any issues with the right lib.

Regards,
Juan A

mlvandenbos, 2019-04-02:
I tried the other library but it also gives no reading.

X = 0.00 mT; Y = 0.00 mT; Z = 0.00 mT

I checked the response code of the 'updateData()' call and it returns '1' so if i'm correct that indicates a bus error?

Should I open an issue on the other repo?

Mark