Infineon/TLE493D-3DMagnetic-Sensor #8: Tle493d-A1 Bare Chip only Reading 0xff

LandLock96, 2019-05-21:
Bare chip tle493d-A1, arduino uno chip picked up on scanner (0x44)
can only receive 0xff from registers , although reset code works.
when using library crash occurs when writing to  mod1 register as writing anything other the 0xE0 causing chip crash, if i try to bypass mod1 register library will receive same 0xff.

letrend, 2019-05-28:
same here using the adafruit huzzah board and a custom pcb with tle493d-w2b6 A0.
i modified the Tle493d.cpp:
line 25: //mInterface.bus->end();
line 35-37: 
        //TURN ON THE SENSOR
	pinMode(14, OUTPUT);
	digitalWrite(14, HIGH);
where the power line of the sensor is connected to gpio14.
The sensors responds to 0x35, the scl line is triggered regularily, but all registers read 0xff

Sync594, 2019-07-22:
I am facing the same issue here. I even tried to run it in STM32 microcontroller, but got similar result. I tested two versions of the sensor - A0 and A1, both of them are giving me just 0xFF. I did several combinations like with and without reset sequence, different configuration settings etc. I can barely turn ON and OFF the INT feature and nothing else. Any help or suggestion is welcome.

letrend, 2019-08-23:
> same here using the adafruit huzzah board and a custom pcb with tle493d-w2b6 A0.
> i modified the Tle493d.cpp:
> line 25: //mInterface.bus->end();
> line 35-37:
> //TURN ON THE SENSOR
> pinMode(14, OUTPUT); [...]

the problem is the 2byte protocol, you need to set the sensor to one byte protocol by writing bit 4 in register 0x11 ( https://forum.arduino.cc/index.php?topic=630711.0 answer 12 )
[image]

closed 2023-04-05 by jaenrig-ifx