Infineon/TLE493D-3DMagnetic-Sensor #17: TLE493d-a2b6 -> Arduino

JugoOriginal, 2021-07-14:
Hi there!
so i bought the "TLE493D magnetic sensor 2go" and followed the steps on this site (https://github.com/Infineon/TLE493D-3DMagnetic-Sensor) to get the data on my serial monitor
After i did that successfully, i designed a pcb board according to the data sheet and installed  the sensor itself on it 
(TLE493D - A2B6 3d Magnetic Sensor Eval Kit   /////  MPN: TLE493D - A2B6 MS2GO).

The board is connected via 4 pins to an Arduino Mega 2560 and the Arduino to my Laptop.

- 1st. pin SCL (21)
- 2nd. pin SDA (20)
- 3rd. pin 3.3V
- 4th. GND

[image]

As you cann see, in the end i want to have 4 of them on my board. BUT first i need to get the data from one of them on my serial monitor in arduino ide.

I tried to use the previous code and library to get the data but that obviously didnt work (so far) and this is the error message:
[image]

because i am completely new to all this stuff i´m not sure if there is a very simple solution to my problem or if its gonna be a big issue.

I am thankful for every help i can get and i already appreciate all the stuff i found so far regarding this topic!

9Volts9er, 2021-07-14:
Hi,
this seems to be the same issue as this one: #16 
I recently did a [commit](https://github.com/Infineon/TLE493D-3DMagnetic-Sensor/commit/4572880d0095f3dc7bc0381517af3d5082148260) with some changes to the repository, that should fix this problem.
Can you maybe check if this solves your issue?

Therefore you would just need to install the latest version of the library by hand:
1) go to [https://github.com/Infineon/TLE493D-3DMagnetic-Sensor](https://github.com/Infineon/TLE493D-3DMagnetic-Sensor) 
2) Click on the green button **`Code`**, then **`Download ZIP`**
3) In Arduino IDE: go to **Sketch > Include Library > Add .ZIP Library** and select the downloaded ZIP file.

JugoOriginal, 2021-07-14:
thanks for the quick reply!
yeah ill try your suggestion

JugoOriginal, 2021-07-14:
Hey 9Volts9er
so i used the updated library and it doesn´t show me any errors anymore which is great! So thank you for that.
But it doesnt update the data either.
[image]

OR what also happened sometimes, is that it only showed me the x-values for the magn. field.
[image]

So then i wasn´t sure if it is my sensor or the code. So i uploaded the code on my test sensor, the "3D magnetic sensor 2go" and i had the same results.

closed 2021-07-15 by JugoOriginal

ozzcet, 2024-03-31:
> Hey 9Volts9er so i used the updated library and it doesn´t show me any errors anymore which is great! So thank you for that. But i only get the x-values from my sensor [image]
> 
> Did something like this happen before?
> 
> Update:
> 
> After changing into "fast mode" i was able to read all of the values! So thank you very very much!!:):)

Where could you change it to "Fast mode" may you express please...

9Volts9er, 2024-04-03:
In the "cartesian low power" example, the sensor is configured into "low power mode".
Simply exchange the "LOWPOWERMODE" with "FASTMODE":

`Tle493d_w2b6 Tle493dMagnetic3DSensor = Tle493d_w2b6(Tle493d::FASTMODE);`

ozzcet, 2024-04-03:
> In the "cartesian low power" example, the sensor is configured into "low power mode". Simply exchange the "LOWPOWERMODE" with "FASTMODE":
> 
> `Tle493d_w2b6 Tle493dMagnetic3DSensor = Tle493d_w2b6(Tle493d::FASTMODE);`

Thank you so much .. it is running for me . Howewer program upload to arduino makes some troubles  while pins connected to arduino. Before upload I disconnected pins to upload succesful.

JugoOriginal, 2024-04-03:
Once i uploaded the code it also happened a lot that the sensor weren´t measuring and simply plugging it in and out will do the trick :)