Infineon/TLE493D-3DMagnetic-Sensor #18: Change Address

JugoOriginal, 2021-07-15:
Hey there!
As i described in #17 i want to work with 4 of the TLE493d-a2b6 simultaneously. So far i designed my pcb board, installed ONE test sensor and i can see the data on my serial monitor in the Arduino IDE using the updated library.

So now i am trying to change the address of that sensor, because when the other ones are installed, i want to communicate with each of them seperately via I^2C and receive my data. 

This turned out to be much more difficult than expected. The code in the library is quite complex (for me at least) and after working on it with multiple people, we still couldn´t figure out how to do it.

So after receiving great help yesterday, i thought maybe somebody else already faced a problem like this and has some advice for me. 

I welcome and appreciate every help i can get!

9Volts9er, 2021-07-16:
Hi JugoOriginal,
ok, so when you're working with multiple sensors on one I²C-bus, I see only two options to realize that:
 - You buy sensors with preconfigured addresses (A0 ... A3). This would probably require that you switch from type A2B6 to W2B6, W2BW or P2B6, as only these types have preconfigured I²C addresses.
  - You implement an electric circuit to turn on each sensor separately (e.g with a MOSFET to each VDD). You start your application with all sensors off. Then you turn on one sensor and change its address to 3. While keeping it running, you then turn on the next sensor and configure it to address 2, then the next to address 1 and then you simply turn on the last sensor as its default address is 0. Then you have all 4 sensors running on one bus.

The Arduino code for the 2nd option should be somewhat similar to this example from the TLV493D-A1B6 library: [DualSensor.ino](https://github.com/Infineon/TLV493D-A1B6-3DMagnetic-Sensor/blob/master/examples/DualSensor/DualSensor.ino)

JugoOriginal, 2021-09-16:
Hi 9Volts9er,
i was busy with other things regarding my study the past weeks but now i am ready to tackle this issue once again.
I ordered now 3 of the TLE493d-W2B6 (A1,A2, and A3) and i kept one TLE493d-A2B6 because it already has A0 as standard address. 

I know that i need the "read" command for each sensor so in total 4 times (or at least i think so) but i do not have an concrete approach for the whole thing. I mean i am an mechanical engineer and had very basic programming and this thing right here seems pretty advanced in my opinion... I really wanna be done with this thing, so i can start my bachelor thesis and finish my study but this one is a really tough one for me.. 
So yeah, once again, any help is hugley appreciated..

9Volts9er, 2021-09-21:
Hi,
I modified the cartesian example to read out 4 sensors. Unfortunately I couldn't test it, because I only have the A0-Type here.
At least it compiles. Can you check if it works with all sensors?

```
#include <Tle493d.h>
#include <Tle493d_a2b6.h>
#include <Tle493d_w2b6.h>

// Four sensors with different I2C addresses
Tle493d_a2b6 Sensor1 = Tle493d_a2b6(Tle493d::MASTERCONTROLLEDMODE, Tle493d::TLE493D_A0);
Tle493d_w2b6 Sensor2 = Tle493d_w2b6(Tle493d::MASTERCONTROLLEDMODE, Tle493d::TLE493D_A1);
Tle493d_w2b6 Sensor3 = Tle493d_w2b6(Tle493d::MASTERCONTROLLEDMODE, Tle493d::TLE493D_A2);
Tle493d_w2b6 Sensor4 = Tle493d_w2b6(Tle493d::MASTERCONTROLLEDMODE, Tle493d::TLE493D_A3);

void setup() {
  Serial.begin(9600);
  while (!Serial);

  // Initialize all sensors
  Sensor1.begin();
  Sensor2.begin();
  Sensor3.begin();
  Sensor4.begin();
}

void loop() {
  // Reading data from all sensors
  Sensor1.updateData();
  Sensor2.updateData();
  Sensor3.updateData();
  Sensor4.updateData();

  // Printing data to console
  Serial.print("Sensor 1\t");
  Serial.print(Sensor1.getX());
  Serial.print(" , ");
  Serial.print(Sensor1.getY());
  Serial.print(" , ");
  Serial.println(Sensor1.getZ());  
  Serial.print("Sensor 2\t");
  Serial.print(Sensor2.getX());
  Serial.print(" , ");
  Serial.print(Sensor2.getY());
  Serial.print(" , ");
  Serial.println(Sensor2.getZ());
  Serial.print("Sensor 3\t");
  Serial.print(Sensor3.getX());
  Serial.print(" , ");
  Serial.print(Sensor3.getY());
  Serial.print(" , ");
  Serial.println(Sensor3.getZ());
  Serial.print("Sensor 4\t");
  Serial.print(Sensor4.getX());
  Serial.print(" , ");
  Serial.print(Sensor4.getY());
  Serial.print(" , ");
  Serial.println(Sensor4.getZ());
  Serial.println();
  
  delay(500);
}
```

JugoOriginal, 2021-09-22:
Hey man,
first of all thank you very much for your help!
So, my attempts went into the same direction (which i´m happy about), but unfortunately with the same result..
[image]

I think that the libraries in line 2 and 3 don´t get included for whatever reason... or what do you think?

9Volts9er, 2021-09-22:
I don't think there is a problem with the includes, otherwise the program wouldn't compile...

For debugging purposes, could you maybe share your schematic with me?
And also, if you have access to an oscilloscope, could you check the SDA- and SCL-line with it? Interesting would be especially the first few bytes, which are sent/received ( at `Sensor1.begin()` ). There we could see, if the sensor reacts properly. It can also be, that there is a problem with the interrupt pulses of the sensors. That would then probably also be visible on the scope.

JugoOriginal, 2021-09-22:
I´ll be on friday in the lab and the boys there have an oscilloscope for sure and then i can check that.
The thing is, it still works if i try to read out only 1 sensor..
[image]

Schematic:
[image]

btw i also noticed that you didn´t use "enable temp". Did you do that on purpose?

9Volts9er, 2021-09-23:
Ah ok, so at least a partial success, when one sensor works :) 

I did another test with my limited possibilities. I changed the address of my sensor to type A1 (0x22) and tried then to use the library to read it with this address. That did not work at first, the sensor always read 0xFF, the same problem you experience.
Digging a bit into the code, I found a bug in the Tle493d.cpp.
There in the function `begin()` the sensor address was configured back to type A0. So it was not possible to read it with address A1 anymore. 
I rearranged the code a little bit as you can see in Pull Request #20 . This Pull Request should solve the issue. 
Can you try, if your setup works with these changes? Would be great to have it properly tested before I merge it.
For this you can simply checkout this branch: [https://github.com/Infineon/TLE493D-3DMagnetic-Sensor/tree/Multiple-sensor-patch](https://github.com/Infineon/TLE493D-3DMagnetic-Sensor/tree/Multiple-sensor-patch)

Thanks a lot!

JugoOriginal, 2021-09-28:
Hey man,
so friday i was in the lab and as i said, we checked with the oscilloscope the hardware and everything looks as it should.
I updated the file with the code you provided and now 2 of 4 (A0 and A1) sensors are working. 
[image]

Which is weird. From my opinion, it should only work with either 1 sensor, or all 4 of them. Because if i am able to communicate with ONE non-standard address (A1, A2 or A3) then i should be able to commucnicate with all of them.

I also tried afterwards to read the values from only the missing sensors (A2 and A3), but that didn´t work either for whatever reason.

9Volts9er, 2021-09-29:
I finally found it :) 

The problem was again here:
```
//correct reset values for other product types
switch (mProductType)
{
  case TLE493D_A1:
  setRegBits(tle493d::IICadr, 0x01);
  break;
  case TLE493D_A2:
  setRegBits(tle493d::IICadr, 0x10);
  break;
  case TLE493D_A3:
  setRegBits(tle493d::IICadr, 0x11);
  break;
  default:
  break;
}
```

The address-bitfield in MOD1 register is just two bits wide and accepts following values: **00**, **01**, **10** and **11**.
So the problem was that the arguments of setRegBits() were given in hexadecimal format instead of binary format.
Therefore A0 did work, because it doesn't set any bits. A1 did work, because 0x01 is equal to 0b01.
But for A2 the function tried to set the bits (in binary) to 0b100**00** instead of just 0b**10**. And for A3 the bits were set (in binary) to 0b100**01** instead of 0b**11**.
Therefore just those two sensors did not work.

Simply changing "**0x**" to "**0b**" fixes this error. 

I added another commit to Pull-Request #20. 
Can you please check if it works now?

JugoOriginal, 2021-09-29:
Hey :)
Look at this!
[image]

Everything works!!!! That´s so freaking cool!!!!!!!!!!
I would´ve never been able to find that mistake or correct it... So thank you sooooo much!

So I have 2 last questions to you:

1. I would like to send you my EAGLE-Files, which is the schematics and the PCB-Board i used, so that what i did is also useful to other people if they have interest in doing the same. I also wrote a Labview-program which shows the data graphically over time, which helps then analyze them. This looks CURRENTLY like this:

[image]

But i am not done yet, because i still need to save the data somehow for further processing.

I think it would be nice to contribute my work to the community and give something back.

So if you would like to upload those things, then sent me your mail address and as soon as i am done you´ll receive everything.

2. Would you like to schedule a zoom-meeting with me, so you could explain to me in detail what you exactly changed to the code and i would take some notes, just for documentation purpose and to include in my presantation? (obviously everything will stay anonymous)
max. 15-30 mins

In any case, thank you again so so much for what you did and commited to this project! It helped a lot:)

9Volts9er, 2021-09-30:
I'm glad that it works now. So I will merge the changes and close this Issue.

You can find my email address in my profile, I just made it visible. Feel free to send me a mail :)

closed 2021-09-30 by 9Volts9er