Infineon/TLE493D-3DMagnetic-Sensor #22: Improving sampling rate in Arduino IDE with multiple sensors

JugoOriginal, 2022-06-29:
Hey there!
Currently i am using 4 different sensors and it is going well so far, but i was asking myself if it is possible to improve the sampling rate..
I think right now it is around 50Hz and it is not as fast as i want it to be. On the datasheet it also says that a rate of 7,8 kHz is possible. Not sure under which circumstances this is achievable but i get the feeling that there is room for improvement with the current rate i have. Or maybe this is just the limit with a serial communication with arduino..
Any kind of help is much appreciated as always!
Thank you very much

[image]

9Volts9er, 2022-06-30:
Hi,
which Serial Baudrate are you using? Have you increased it to something like 115200?
I guess the `Serial.print()` commands would otherwise also take quite some time.

JugoOriginal, 2022-06-30:
Hey man!
It is you again :)
Yeah i already read about it and also increased it but yeah it doesn´t really work..
[image]
[image]

ArkajyotiChatterjee, 2022-06-30:
In your second image, the Baud Rate is not updated to 115200.
Please do that and try again.

[image]

Click the Drop down and select 115200.

JugoOriginal, 2022-06-30:
yeah i already saw that and tried that as well but then just nothing happened
[image]

i just thought, i might as well try to change it in the code but that didnt work either

9Volts9er, 2022-06-30:
Did you try to change it in the code AND in the Serial Monitor window? ;)
In the code you set the Baudrate in the microcontroller. In the Serial Monitor you say, which Baudrate is expected by the PC. Only if those two values match, it will work fine.
Sorry if that was already clear to you, but it sounded like you tried to change only either the value in the code or the value in the Serial Monitor - and not both...
If you did that and you still do not get any output in the monitor, you could try to remove the `while(!Serial);`.

JugoOriginal, 2022-06-30:
Oh yeeaaahhh i didn´t see that!
I changed only one of them -.-
No no please don´t apologize. What i meant was, that i was aware of the baud rate in the bottom right corner of the serial monitor and i also changed it, but not in combination with the code.. 
Here is the improvement:

So it went from this

[image]

to this

[image]

So it went from ≈15Hz to ≈120Hz!

This is a great improvement! I also removed the **_while(!Serial);_** and i tried all the baud rates until 2000000 but neither of those things impacted the rate in any way.

Anywas, this already helps me A LOT so thank you very much to the both of you!!

Is there anything else coming to your mind that i can try?

9Volts9er, 2022-06-30:
I'm glad to hear, that this helped. 

Ok, let's see. If you haven't changed anything yet, the standard I2C clock speed of your Arduino Mega should be 100 kHz.
The library always reads 7 registers + 1 address byte with each 9 bits (incl. ACK) per readout of one sensor. So for 4 sensors you ideally would be limited to 100000/(8*9*4) => ~350Hz. 
Of course using the library is not time efficient either, it's designed to be easy to use but not to be very fast. So this might also be a reason for the lower speed.
And I also don't know how precise the timestamps in the Serial Monitor are. If you have access to an oscilloscope you could check how long your I2C lines are busy when reading the sensors.

So you could try two more things:
- Increase the I2C clock speed
- Using raw I2C commands to readout the sensors. You can use [this example](https://github.com/Infineon/TLI493D-W2BW/blob/master/examples/Raw_I2C_readout/Raw_I2C_readout.ino) of the TLI493D-W2BW as reference. Please note that you will need to change the settings in MOD1 register to disable INT and CA (Also the IICadr bits need to match your addresses and the parity bit needs to be set accordingly). The register contents are described in the [User Manual](https://www.infineon.com/dgdl/Infineon-TLE493D_P2B6-UserManual-v01_00-EN.pdf?fileId=5546d4627862c3e5017868939a34560a).

In the library there is no `delay()` that would decrease the reading speed. So I think this are your best chances of improvement.

JugoOriginal, 2022-06-30:
Yeah i am super happy about the result!

To be honest, i don´t really know how to do the things that you suggested. I am an mechanical engineering student and this already goes way beyond what i learned in uni. So i guess I´ll just take the 120Hz and be happy with it :).

Just one last question before i close the issue, is it possible to display the values with a comma instead of a point. For example instead of "12.34" it would show me then "12,34". Not sure if it is possible, because the way numbers are displayed with a data type is kind of a constant. Or am i wrong?

9Volts9er, 2022-07-01:
Well, actually to display those values in the Serial Monitor they first get converted to a string.
You can simply go to the function that does this for double and float numbers and replace the '.' by an comma. That would be the easiest way, I guess.
You should find this function by opening the following folder in the Windows Explorer:
`C:\Users\%USERNAME%\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.4\cores\arduino`
Then you open the `Print.cpp` file and go to the function `size_t Print::printFloat(double number, uint8_t digits)`. In my case this was in line 223.
There you go down until you find the following lines (for me the lines 251-254):
```
  // Print the decimal point, but only if there are digits beyond
  if (digits > 0) {
    n += print('.'); 
  }
```
And there you replace the '.' by an ','. Then you should get values printed to the Serial Monitor separated by comma instead of point.

And if you are that far, then you can also change your I2C default clock speed in a similar way.
Go to the file location:
`C:\Users\%USERNAME%\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.4\libraries\Wire\src\utility`
Open the `twi.h`, and change the `#define TWI_FREQ 100000L` to `  #define TWI_FREQ 400000L`.

JugoOriginal, 2022-07-01:
1. so i found the file but it was stored in a different place than yours

[image]

Changing the point to a comma didn´t do the trick unfortunately

[image]

[image]

2. the other file you told me to look for doesn´t exist on my laptop. I used also the search function in the explorer but i didn´t get any results

[image]

3. I´ll just ask you one last question, is there a possibility to display more than 2 decimal places?

It is not so important to change the point to a comma so don´t spend too much time on the issue. You already helped me a lot which i am very thankful for.

9Volts9er, 2022-07-01:
This is the file location for Infineon Microcontrollers (XMC). You are using the Arduino Mega, aren't you?
So this will for sure not work then.
And you are then probably not using the same Version of the IDE as I do. Or are you maybe using the Windows App?

You could try to install the newest Version of the IDE. Then go to File->Preferences and below the "Additional Boards Manager URLs" you should find "_More preferences can be edited directly in the file_" and then a File location (should be a clickable link). If you click on this link or navigate to the folder in the Explorer, then you should be in your base folder of the Arduino files. 
Go to "packages->arduino->hardware->avr" then the version number (for me "1.8.4").
Under "cores->arduino" you will find the `Print.cpp`, and under "libraries->Wire->src->utility" you will find the `twi.h`

At least this was the case for me. But if the files are not located there, then you might just google the right file location of the Arduino AVR files with your version of the IDE :)

To 3): just use `Serial.print(x, 5);` to display the variable `x` with 5 decimal places.

JugoOriginal, 2022-07-01:
I just tried a few things and also your approach via the preferences but i just kept landing in the same folder where i found the print.cpp file from before. I also jus tried creating the twi.h file with the code i found on github but that didn´t do anything either.. 
Also installing Arduino again on a different laptop didn´t bring the wanted result. What a mystery this is..

If I change to 4 decimal places, it just shows me more zeros as you can see here 

[image]

And just one thing purely out of curiosity, why do i keep seeing  "0.13" and "0.26" ?

9Volts9er, 2022-07-04:
I'm not sure then, where the Arduino folder is for your PC. The Print.cpp you found before was in the "Infineon" folder, this does only affect the XMC microcontroller.
Maybe you find it in the folder `C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino`. But I'm not sure if changes have any effect here, you can try it.

To get more decimal places and a better resolution, you would need to do some averaging. So read for example 100 values and average over them. Then you will get a more precise value with a higher resolution.

0.13 is 1 LSB of the sensor and 0.26 is 2 LSB. So if you have no magnetic field applied, this is just noise. When you use the averaging method, you will decrease the noise as well.

JugoOriginal, 2022-07-11:
Hey!
yeah i solved the problem.. it was quite easy to be honest.. i just had to update the board and the correct file was available!
[image]
I changed the `#define TWI_FREQ 100000L` to `#define TWI_FREQ 400000L and my sample rate was improved to ≈165Hz which is an improvement of roughly 33%!

So i am more than pleased with this result and i really appreciate it that you took the time to help me with this!
Thank you very much!

closed 2022-07-11 by JugoOriginal