Infineon/TLE493D-3DMagnetic-Sensor #24: Bus error

harry-finch, 2023-05-09:
Hi,

I have looked on this forum and on the Arduino forum, but for the life of me I can't figure out what's wrong...

I'm using a bare metal TLE493D-W2B6 A3 and have wired everything up on my breadboard with an Arduino Uno. I'm using 3V3 and have added 1K pull-up resistors as well.

I have uploaded the cartesian.ino example just to test everything but I always get the same values for all 3 axis (-266.24) even when I get a magnet real close. I have checked the value returned by the updateData function and I get 1. Which means I have a bus error. However I don't really know what to do with this...

I have used the i2c scanner to check the device's address and it returns 0x44 alright.

It looks like it's working for a couple of seconds at a time (the value returned becomes 1 and the values are closer to 0), I don't know why, I'm not doing anything... This is really puzzling to me.

I'm using the Arduino IDE 2.0.3 on MacOS.

All help is welcome!

9Volts9er, 2023-05-10:
Hi,
so you get some values, but after a few seconds the Arduino only reads 0xFF, right?
That's strange. If it never worked, I would have guessed, that after turn on, the sensor is not configured fast enough, so it sends an Interrupt-pulse on the SCL line, which could have lead to a hangup of the I²C hardware of the Arduino. I've had this problem a few times, and it can be solved by controlling the sensors VDD via a transistor to switch it on and immediately configure it.

However as you can read a few values correctly, it means the library has already disabled the INT pulse, because it gets deactivated by default within the `sensor.begin()`.

So my next guess would be that there is some problem with the voltage levels. The Arduino is a 5V device, and the sensor is a 3.3V device. That should work in theory thought, because if the voltage rises above 3V the Arduino detects it as HIGH. 
But maybe it rises there too slowly because of some parasitic capacities or something.
Or the fall time of the signal is too long for the sensor.
Do you have the possibility to measure the SCL and SDA lines with an oscilloscope? That could help

harry-finch, 2023-05-11:
Ok, scratch that, I'm using an A0 not an A3 with the Arduino so he was trying to get 0x44 when the address is actually 0x35. The result is still weird though...

[image]
[image]

There is some kind of a response now, the line is pulled to ground, but it doesn't mean anything.

9Volts9er, 2023-05-12:
It seems that the sensor is not responding to the address 0x35 (NACK).
The short LOW pulse is the Interrupt-pulse from the sensor. This shows at least, that the sensor is up and running, it sends this pulse after every ADC sampling cycle. But this shows also that the sensor it not configured by the library. Otherwise this pulse would be deactivated.
Are you sure, you have the A0 type? Because in the first picture, it seems like there is no INT pulse, so that would mean, the sensor is configured properly. Also the picture is zoomed out very far, so maybe we just don't see the INT pulse.

Can you try the following code? There might be bugs in it, I have not tested it, but it is a very raw code using only I2C commands, no library code is involved at all. Therefore it should be easy to debug as every byte sent via I2C can be seen in the code as well. 
You can test all 4 addresses: 0x35, 0x22, 0x78 and 0x44. On the I2C lines you should then see for example if you use address 0x44 the following sequence: 0x00, 0x00, 0x44+W, 0x10, 0x11, 0xD5 and then periodically 0x44+R, followed by 7 bytes of sensor data.

```
#include <Wire.h>       // default I²C library

#define ADDRESS 0x44

#define DELAY 100       // value in ms, change to modify update rate

void setup() {
  // Init I2C peripheral
  Wire.begin();

  // Reset sensor
  Wire.beginTransmission(0x00);
  Wire.write(0x00);
  Wire.endTransmission();

  // Configure sensor
  Wire.beginTransmission(ADDRESS); // Sensor address
  Wire.write(0x10);             // Register address
  Wire.write(0b00010001);       // Config register: measure all channels, full range, ADC trigger on read before first MSB, odd CP parity

  if(ADDRESS == 0x35)
  {
    Wire.write(0b00010101);       // MOD1 register: master controlled mode, one byte read command, enabled clock streching, no interrupt, odd FP parity
  }
  else if(ADDRESS == 0x22)
  {
    Wire.write(0b10110101);       // MOD1 register: master controlled mode, one byte read command, enabled clock streching, no interrupt, odd FP parity
  }
  else if(ADDRESS == 0x78)
  {
    Wire.write(0b11010101);       // MOD1 register: master controlled mode, one byte read command, enabled clock streching, no interrupt, odd FP parity
  }
  else if(ADDRESS == 0x44)
  {
    Wire.write(0b01110101);       // MOD1 register: master controlled mode, one byte read command, enabled clock streching, no interrupt, odd FP parity
  }
  Wire.endTransmission();
  
  // Init serial port for communication to the PC
  Serial.begin(9600);
}

void loop() {
  // Readout the first 7 data bytes
  uint8_t buf[7];
  Wire.requestFrom(ADDRESS, 7);
  for (uint8_t i = 0; i < 7; i++) {
    buf[i] = Wire.read();
  }

  // Built 12 bit data 
  int16_t X = (int16_t)((buf[0] << 8) | (buf[4] & 0xF0)) >> 4;
  int16_t Y = (int16_t)((buf[1] << 8) | ((buf[4] & 0x0F) << 4)) >> 4;
  int16_t Z = (int16_t)((buf[2] << 8) | ((buf[5] & 0x0F) << 4)) >> 4;
  uint16_t T = (buf[3] << 4) | (buf[5] >> 4);
  
  // Send via serial port to be displayed on a terminal 
  Serial.print(X);
  Serial.print(" ");
  Serial.print(Y);
  Serial.print(" ");
  Serial.print(Z);
  Serial.print(" ");
  Serial.print(T);
  Serial.println();

  delay(DELAY);
}
```

harry-finch, 2023-05-12:
> Are you sure, you have the A0 type? Because in the first picture, it seems like there is no INT pulse, so that would mean, the sensor is configured properly. Also the picture is zoomed out very far, so maybe we just don't see the INT pulse.

To be honest, I'm not sure of anything... The only thing I can tell is that when I used 0x44 I got absolutely nothing (screenshots from my first rreply) but when I used the I2C scanner it told me there was a device with 0x35 address. Then with that address I got the other screenshots you're talking about. The second screenshot is a zoom in from the first.

Alright, we might be getting somewhere... Here is what I got from your code. It looks like the address is 0x44 after all.

[image]
[image]
[image]
[image]

The results printed out with serial for 0x44 is `31 -1 -1 4095` and only the X value is changing when I get closer with a magnet. Which is already something.

9Volts9er, 2023-05-16:
Ok, yes, the address 0x44 is correct then. At least, we have that.
Maybe the enabled clock streching is the problem, I'm not sure if the Arduino can handle this properly...
Can you try a different configuration:

```
#include <Wire.h>       // default I²C library

#define ADDRESS 0x44

#define DELAY 100       // value in ms, change to modify update rate

void setup() {
  // Init I2C peripheral
  Wire.begin();

  // Reset sensor
  Wire.beginTransmission(0x00);
  Wire.write(0x00);
  Wire.endTransmission();

  // Configure sensor
  Wire.beginTransmission(ADDRESS); // Sensor address
  Wire.write(0x10);             // Register address
  Wire.write(0b00100001);       // Config register: measure all channels, full range, ADC trigger after read of 0x05, odd CP parity

  Wire.write(0b11111101);       // MOD1 register: master controlled mode, one byte read command, disabled clock streching, no interrupt, odd FP parity

  Wire.endTransmission();
  
  // Init serial port for communication to the PC
  Serial.begin(9600);
}

void loop() {
  // Readout the first 7 data bytes
  uint8_t buf[7];
  Wire.requestFrom(ADDRESS, 7);
  for (uint8_t i = 0; i < 7; i++) {
    buf[i] = Wire.read();
  }

  // Built 12 bit data 
  int16_t X = (int16_t)((buf[0] << 8) | (buf[4] & 0xF0)) >> 4;
  int16_t Y = (int16_t)((buf[1] << 8) | ((buf[4] & 0x0F) << 4)) >> 4;
  int16_t Z = (int16_t)((buf[2] << 8) | ((buf[5] & 0x0F) << 4)) >> 4;
  uint16_t T = (buf[3] << 4) | (buf[5] >> 4);
  
  // Send via serial port to be displayed on a terminal 
  Serial.print(X);
  Serial.print(" ");
  Serial.print(Y);
  Serial.print(" ");
  Serial.print(Z);
  Serial.print(" ");
  Serial.print(T);
  Serial.println();

  delay(DELAY);
}
```

harry-finch, 2023-05-16:
Oh my God, it's working!! Thank you!

So it was the clock stretching? Is it something we can configure with the library?

harry-finch, 2023-05-16:
I made the example sketch work by adding `Tle493dMagnetic3DSensor.disableInterrupt();` and `Tle493dMagnetic3DSensor.disableCollisionAvoidance();` in the setup().

Many thanks for your help!

closed 2023-05-16 by harry-finch

9Volts9er, 2023-05-17:
Yes, just be aware, that you are now working with an unsynchronized readout. The library sets the TRIG bits by default to 0b01, which means the sensor starts sampling as soon as you readout register 0x00.
And for example when you read out the y MSB, the sensor might be done with sampling the new y value and updates the registers inclusive the y LSB register. Therefore the MSB and LSB might not match, because they are from different sensor sampling times.
You can prevent this by setting the TRIG bits to 0b10 or 0b11.

This can be done here in the `Tle493d.cpp` [line 95](https://github.com/Infineon/TLE493D-3DMagnetic-Sensor/blob/6b18a2641df5f257e8e4898b454be89e41d910e2/src/Tle493d.cpp#LL95C3-L95C3):
`setRegBits(tle493d::TRIG, 2);`

Now you only have the problem that you basically lag one readout behind the actual magnetic field values.
The order is like this:
1. You readout the sensor values
2. The sensor samples directly after you finished the readout
3. You probably wait for some time until the next readout, e.g. delay 100ms
4. You readout the sensor values and get the magnetic field values from 100ms ago

Therefore you might want to readout the sensor values twice in the `loop()` before working with them:
```
Tle493dMagnetic3DSensor.updateData(); //Trigger sensor readout
delay(1); //Wait until conversion is finished. Delay could be reduced even further if necessary.
Tle493dMagnetic3DSensor.updateData(); //Read sensor values from only 1ms ago
...
```

harry-finch, 2023-05-21:
Alright, many thanks for your help! Just out of curiosity, do you have any idea why clock stretching isn't working with my chips?

9Volts9er, 2023-05-22:
Clock stretching is an optional feature in the I²C protocol. So it's not implemented in all devices.
My guess is, that the Arduino Uno (ATmega328P) is one of those microcontrollers that doesn't support clock stretching, but I've not verified that. Maybe there is a register setting that enables it.