Infineon/TLE493D-3DMagnetic-Sensor #27: MS2GO keeps giving a wrong XYZ-values (-266.24) from serial monitor

LUKEDUDE97, 2023-08-22:
I have noticed there are some similar cloased issues yet no detailed solution has been provided. Like it happened to #25 , the poster seems don't know the answer either with the bug just disappear itself. But for me, -266.24 just keeps appear in my serial monitor. 

[image]

I am using a tle493d w2b6 MS2GO, arduino (2.1.1), TLE493D library(1.3.0) and XMC 2.2.0

I presume it may relate to J-link or board itself, because there's only one time it goes well with right XYZ-output when i reinstall the J-link, update it from an old version. But it soon broke after i reupload the code. I aslo tried FASTMODE and every possible operations from other cloased issues but it is just still broken. 

I'm totally noob on this stuff, hoping someone can give me some advice. Thank you very much!

###### THANKS!

LUKEDUDE97, 2023-08-22:
I downgrade my TLE493D library to 1.2.0 and the error disappear permannently. I wonder why. 

That one-time right running after i updated my J-link makes me really currious what happened. It just struggles me till now. I still hope i can fix the problem under 1.3.0 library.

LUKEDUDE97, 2023-08-22:
Yeah, similar issue #16 happened under 1.2.0 when i tried to connect my arduino nano with the sensor board. 

[image]

Even thougth it seems all right when directly wire my PC with the MS2GO under 1.2.0, but still, it turns out i still have to go back to 1.3.0 and the error still remains......

LUKEDUDE97, 2023-08-22:
It backs to right under 1.3.0 after i replug the wire to board. But sometimes it randomly goes back to wrong. It just bizzar....And sometime there's even no serial output from arduino. Quite count on luck...

And i'm still wondering why it goes wrong under 1.2.0 when directly wire my PC to MS2GO. That -266.24 is just annoying.

LUKEDUDE97, 2023-08-22:
It backs to right under 1.3.0 after i replug the wire to board. But sometimes it randomly goes back to wrong. It just bizzar....And sometime there's even no serial output from arduino. Quite count on luck...

So all in all, 1.3.0 would cause error when the MS2GO is directly connected to the pc under my context at least. And it is still seemingly unstable when connect it via an Arduino board.

9Volts9er, 2023-08-23:
Hi,
can you share the Arduino Code you are using?
If you are using the "Cartesian" example, have you enabled the lines 17-19 in order to power on the sensor?

LUKEDUDE97, 2023-08-23:
Oh. How can i miss that inline notice after all trying...... OK, now it turns fine under 1.3.0. THANK YOU

closed 2023-08-23 by LUKEDUDE97