Infineon/TLE493D-3DMagnetic-Sensor #28: -266.24

djqwhqwe, 2023-08-23:
Hey. Can you help? in Cartesian.ino example error after uncommenting lines 

//If using the MS2Go-Kit: Enable following lines to switch on the sensor
pinMode(LED2, OUTPUT);
digitalWrite(LED2, HIGH);
delay(50);

code does not working

Error:
pinMode(LED2, OUTPUT);
            ^~~~
            FE2

exit status 1

Compilation error: 'LED2' was not declared in this scope

What should I do?

9Volts9er, 2023-08-24:
Have you selected the XMC2Go as Board?
[image]

djqwhqwe, 2023-08-24:
Thanks, this helped.

Could you help in the following situation.

I am using arduino mega 2560 and ramps 1.4.
Connecting the 2go sensor uisng ramps i2c , I get -266.24 -266.24 -266.24

Can you tell me what can be done with this, I cannot set the pin '+' to the output

[image]

9Volts9er, 2023-08-25:
How did you connect the 2Go sensor? So which pin of the 2Go board was connected with which pin of the RAMPS board?
Is the XMC on the 2Go board still connected, or did you cut the pcb?
As far as I know, the Arduino mega 2560 is running at 5V, the sensor runs on 3.3V. Therefore you need to take care of the voltage difference. For example you can supply the sensor with the 3.3V-pin from the Arduino and connect the I²C pullups (on SDA/SCL) to this 3.3V as well. But you have to make sure, that there are no pullups on the ramps board or the Arduino that would pull the lines to 5V.

Out of curiosity: For what are you using the sensor here?

djqwhqwe, 2023-08-25:
The sensor is connected to the i2c ramps bus. Nothing is cut. In the figure above, i2c is shown in the upper right corner. On the ramps i2c "+" has 5V. I found a line on the ramps that goes to arduino pin 15, sometimes it works, sometimes it doesn't. Looking for a solution. Very thankful for your responses. I am doing a scientific project, but unfortunately I can not disclose the details.

9Volts9er, 2023-08-25:
I just had a look into the schematics of the Ramps 1.4.
[image]
In order to use a 3.3V I²C communication, you need to remove R21 and R22. Those would otherwise pull the SDA/SCL lines to 5V.
For the sensor supply, you can use the 3.3V pin of the Arduino Mega, just connect it via cable to the sensor board here (not to the 3.3V pin, otherwise the XMC will get powered on as well):
[image]

**The sensor will get powered immediately when you supply the Arduino, so make sure the sensor.begin() function in the library is called as soon as possible in your code!**

SCL, SDA and GND can be connected from the I²C pins on the RAMPS board that you've pointed out.
I've not checked if there are external or internal pullups on the Arduino Mega, I don't think so, but maybe you can ensure that yourself.

Hope this helps.

djqwhqwe, 2023-08-25:
I am very grateful for your help. Thank you!

closed 2023-08-25 by djqwhqwe