Infineon/TLE493D-3DMagnetic-Sensor #29: I2C communication with sensor does not work

ibisek, 2023-08-24:
Hello there!

I got a TLE439DA2B6HTSA1 from Mouser (presuming type A2?) currently connected to Bluepill (stm32f103c8; earlier to ESP8266) and I am (like others) experiencing severe I2C communication issues.

When running the I2C scanner I get immediately after reset

Scanning...
I2C device found at address 0x35  !
done

Scanning...
No I2C devices found

.. the consequent scans are not showing anything anymore. Looks like the sensor (or i2c?) always gets stuck.

The address discovered indicates the device is TLE493D_A0 and not A2 (address 0x78) .. or is it some other device type/kind?

Went already through multiple issues (and solutions) here on github but with no success. For example, the code given in [https://github.com/Infineon/TLE493D-3DMagnetic-Sensor/issues/24] yields

-1 -1 -1 4095

when using sensor address 0x35 (dev type A0) and when the address is 0x78 (A2) the printout is

176 10 0 0

When looking at the SDA line with a scope there is data only for the first transaction and then the i2c communication never proceeds. Same situation with bluepill and esp8226 (hence it is probably not mcu related).

The SCL & SDA lines are pulled up by a 1k2 resistors and there is a 100pF cap in place of C4 as recommended here [https://www.infineon.com/dgdl/Infineon-Infineon-3DMS2GO_TLE493D-A2B6-UM-v01_02-EN-UserManual-v01_02-EN.pdf?fileId=5546d462636cc8fb01641834364e3f56] in Figure 9. But the cap has no effect on the comms..

Initially I used the libs downloaded from platformio (infineon/TLE493D-Magnetic-Sensor @ ^1.3.0), now got the latest libraries from the github repo.

What am I doing wrong? Any ideas?

Many thanks for your insights!

ibisek, 2023-08-24:
UPDATE: Did the same as suggested in [https://github.com/Infineon/TLE493D-3DMagnetic-Sensor/issues/17] with Arduino Uno, getting

-266.24 ; -266.24 ; -266.24

as output.

9Volts9er, 2023-08-25:
Hi there,
the sensor you are using is the TLE493D-A2B6 with the I²C address of A0, so the address 0x35 is correct.

I have never personally tried to use this sensor with this hardware, so I can only make assumptions.
Are you able to switch on/off the VDD of the sensor? This would make it a bit easier, because then you can control the point in time when the sensor starts and can reconfigure it directly afterwards.
With the microcontrollers I tried so far, the INT pulse was always a problem. When a INT pulse occurred after the I²C unit of the microcontroller was active, it got stuck and no further communication was possible. Therefore I recommend the following sequence: 

1. Make sure the sensor is off (VDD is not supplied)
2. Configure the I²C hardware unit of the microcontroller
3. Startup the sensor
4. Configure the sensor to One-Byte-Read and switch off the INT pulse.

If you are using the library, the begin() function will take care of all these steps except for powering the sensor. In that case you should set the sensors VDD high right before you call the begin().

ibisek, 2023-08-28:
Hey @9Volts9er!

Thanks for pointing me the right direction!! 

One of the missing bits was definitely the immediate chip configuration after power-up. Even a slight delay is causing the interrupt on SCL pin to kick in and ruin everything.

`// power-cycle the sensor:
  pinMode(SENSOR_PWR, OUTPUT);
  digitalWrite(SENSOR_PWR, 0); delay(200); digitalWrite(SENSOR_PWR, 1); 
// init the sensor (with 0 delay!)
  magSensor.begin();`

However, there still was a trouble in the consequent communication. The init/begin went through, the initial values were read from the registers 0x00-0x07 but were all the time the same. Even on the scope there was no apparent communication, just +3.3V on the SDA and silence. Hence I suspected the 47 ohm resistors need to be in series with SCL&SDA lines. But their presence causes no change (now it works even without those two).

[image]

The next trouble seemed to be in (arduino's) Wire.endTransmission(true) while it must be false (or default; without argument).

And the consequent trouble seemed to be in the library default construstor. It takes form of
`Tle493d_a2b6 magSensor = Tle493d_a2b6();
or
Tle493d magSensor = Tle493d();
`
assuming the access mode to be MASTERCONTROLLEDMODE (which does not work), while using

`Tle493d magSensor = Tle493d(Tle493d::AccessMode_e::FASTMODE, Tle493d::TypeAddress_e::TLE493D_A0);`

does the job! It works now!! :) 

I hope this will help future generations of devs to help them to make the TLE493D-A2B6 magnetic sensor work :)

And for information, this is my current setup:
[image]

Cheers,

Ibisek

closed 2023-08-28 by ibisek