Infineon/TLE493D-3DMagnetic-Sensor #30: Mega 2560 with -266.24

NOWHEREINWUHAN, 2023-09-24:
Sorry to bother here, but -266.24 just keeps appear in my serial monitor, too. (I have looked up other similar closed issues.) My self-design PCB use two 1K2 Ohm pull-up resisters and one 100nF capacitance. Mega 2560 is my miu-C, however I notice that its pins 20(SDA) and 21(SCL) already integrated with pull-up resisters whose exact value or the voltage be pulled-up to I can't find. I try to remove the two 1K2 Ohm, but also didn't work. I wonder what other problems it may be? (My sensor is TLE493D-W2B6-A3-HTSAI)

NOWHEREINWUHAN, 2023-09-27:
I wrote my own Arduino code based on the register map and read it successfully and found that the Mega2560 does not work with INT so clock extension must be used! 
But there is still a problem, two of the sensors A1A2 can be read normally and the other two A0A3 cannot be read, I would like to know if there is a reason for this?

9Volts9er, 2023-10-06:
Hi,
so you are using a setup with 4 sensors? Can you share the schematic?
And you have written your own code? Can you share that too?

NOWHEREINWUHAN, 2023-10-06:
阙子歆同志正在回复您的路上一路狂奔…… 【自动回复】

ederjc, 2024-07-26:
Closed for inactivity. Please re-open [here](https://github.com/Infineon/arduino-xensiv-3d-magnetic-sensor-tlx493d/issues) if it's still relevant.

closed 2024-07-26 by ederjc