Infineon/TLE493D-3DMagnetic-Sensor #32: PWM output

ozzcet, 2024-02-12:
Actually I am searching how could set PWM out according to x,y,z direction movement of magnet. I use platform io with XMC 2Go Kit (actually my board contain TLE493D on board. TLE493D-p2b6-2go) after some test I understand that  the pin of

9Volts9er, 2024-04-03:
1. Configure the sensor (as basis you can use e.g. the "Cartesian" example)
2. In the loop: Readout the sensor
3. Map magnetic field values to PWM output (convert magnetic field values to 0-255 integer values for pwm output)
4. Write PWM to the pins

I've not tested it, but it should work somehow like this:
```
#include <Tle493d.h>

#define MAX_MAG_FIELD 50
#define PWM_X_PIN 1
#define PWM_Y_PIN 2
#define PWM_Z_PIN 3

Tle493d Tle493dMagnetic3DSensor = Tle493d();

void setup() {
  Serial.begin(9600);
  while (!Serial);
  
  //If using the MS2Go-Kit: Enable following lines to switch on the sensor
  // ***
  pinMode(LED2, OUTPUT);
  digitalWrite(LED2, HIGH);
  delay(50);
  // ***
  
  Tle493dMagnetic3DSensor.begin();
  Tle493dMagnetic3DSensor.enableTemp();
}

void loop() {
  double x,y,z;
  //Read magnetic values into variables
  Tle493dMagnetic3DSensor.updateData();
  x=abs(Tle493dMagnetic3DSensor.getX());
  y=abs(Tle493dMagnetic3DSensor.getY());
  z=abs(Tle493dMagnetic3DSensor.getZ());

  //map magnetic field to pwm value
  int pwm_x = map(x,0,MAX_MAG_FIELD,0,255);
  int pwm_y = map(y,0,MAX_MAG_FIELD,0,255);
  int pwm_z = map(z,0,MAX_MAG_FIELD,0,255);

  //write pwm value to outputs
  analogWrite(PWM_X_PIN, pwm_x);
  analogWrite(PWM_Y_PIN, pwm_y);
  analogWrite(PWM_Z_PIN, pwm_z);
  
  delay(5);
}
```

ozzcet, 2024-04-03:
Thank You so much my friend.

closed 2024-04-03 by ozzcet